jcorporation/myMPDos #1: Release an image for armv7 and armv6

m040601, 2020-12-21:
Thanks for your awesome work on myMpd and myMpdos

> myMPDos is a Raspberry Pi image **(aarch64)** 

This means Raspberry Pi models prior to the Rpi3 are not supported.
That leaves a lot of Rpi2 and Rpi zeros behind.

I ran myMpd on a perfectly capable Rpi2 1.2 (does not support aarch64).

Would it possible to release for these archs ?
Automate it through github actions ?

Thanks in advance.

jcorporation, 2020-12-21:
This should be possible. The build script could be enhanced. But I have no devices for testing.

jcorporation, 2020-12-21:
I adapted the build script to respect the ARCH environment variable. Creating the build image works, but booting the images for armv7 and armhf does not work. It seems the kernels for armv7 and armhf has no support of the virtual pci devices.

The build image uses the netboot boot files and the virt machine.

m040601, 2022-03-07:
> booting the images for armv7 and armhf does not work.

>  It seems the kernels for armv7 and armhf has no support of the virtual pci devices.

Could you give an update on the current situation ?

jcorporation, 2022-03-07:
I use the alpine netboot images to compile the binaries with qemu, but the images for older arm architectures are not booting with qemu. 

Workaround would be to compile with docker buildx. This involves reengineering the compile stage of the build script.

m040601, 2023-04-10:
Could you provide any updates on this situation ?

I checked and saw that, https://pkgs.alpinelinux.org/packages?name=snapcast&branch=edge&repo=&arch=&maintainer=

 Alpine linux itself provides packages for both armv7h (raspberry 2) and even armhf (raspberry zero)

Is there still any hope to any mympdos releases for these two ?

closed 2024-04-20 by jcorporation