jcorporation/myMPDos #2: NFS not supported

bbodinier, 2021-01-03:
Thanks you very much for myMPD and myMPDos. It's a very good work, very good idea, very innovative.
I think it would be perfect to integrate NFS at myMPDos.
Otherwise an apk is enough : `apk update` and `apk add nfs-utils`

jcorporation, 2021-01-03:
Thanks for your feedback.

cifs-utils and nfs-utils are by design not installed. I want the number of installes packages as small as posible.

For cifs there is a description how to add it. I enhanced this article and add a description for nfs. Do you use any special nfs options?

Wiki: https://github.com/jcorporation/myMPDos/wiki/Mount-a-CIFS-or-NFS-share

bbodinier, 2021-01-03:
the wiki helps a lot, it's a little different from a system like Debian. The options of nfs are sufficient for my use. myMPDos surpasses archphile musically, so myMPDos uses the latest versions. Thank you !

jcorporation, 2021-01-03:
Right, Alpine is somewhat different from standard distributions, but I think perfect for a raspberry - small size and read only sd-card.

closed 2021-01-03 by jcorporation