jcorporation/myMPDos #3: Wrong mixer_control

bbodinier, 2021-01-04:
Hi,
I have a Khadas tone board, and `/etc/mpd.conf`, in `audio_output` section, `mixer_control` is wrong :
```
audio_output {
  name          "Control USB Audio"
  device        "hw:0,0"
  type          "alsa"
  auto_resample "no"
  auto_format   "no"
  mixer_type    "hardware"
  mixer_device  "hw:0"
  mixer_control "Khadas"
}
```
`#amixer`
```
Simple mixer control 'Khadas Tone Control ',0
  Capabilities: pvolume pswitch
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 127
  Mono:
  Front Left: Playback 107 [84%] [-20.00dB] [on]
  Front Right: Playback 107 [84%] [-20.00dB] [on]
Simple mixer control 'Khadas Tone Control ',1
  Capabilities: pvolume pvolume-joined pswitch pswitch-joined
  Playback channels: Mono
  Limits: Playback 0 - 127
  Mono: Playback 127 [100%] [0.00dB] [on]
```

I tried to change `mixer_control` to the correct value (`mixer_control "Khadas Tone Control "`), but after a reboot, `mixer_control` always reverts to the wrong value. I checked `/media/mmcblk0p2/mpd.apkovl.tar.gz`, `/etc/mpd.conf` file is correct after doing:

```
vi /etc/mpd.conf (edit with correct value)
rwdata.sh
lbu include /etc/mpd.conf
lbu commit -d
```

`/etc/mpd.conf` always reverts to the wrong value.
Thanks

jcorporation, 2021-01-04:
MPD output autoconfiguration is the problem. I must look in the code, but commenting the line with configmpd.sh in /etc/mdev.conf should prevent the overwrite on each reboot.

bbodinier, 2021-01-04:
Commenting the line with `configmpd.sh` in `/etc/mdev.conf` does not work (alsa failed).
I found another solution :
change `ENABLE_MIXER` to `false` in `/etc/mympdos/mympdos.conf` and place the good `mpd.conf` in `/etc/mympdos/custom/`

bbodinier, 2021-01-05:
I agree, autoconfiguration is a problem, it slows down the boot a lot. If it was an option in bootstrap.txt I would leave it set to false because I don't want a script to modify my mpd.conf file.

jcorporation, 2021-01-06:
You can set `ENABLE_CONFIGMPD="false"` in the bootstrap.txt to disable the autoconfiguration of MPD.

Further above is my error, you should only comment out the last column with configmpd.sh not the complete line.

jcorporation, 2021-01-06:
I looked at the configmpd.sh script. The problem is the handling of spaces in the mixer name. I build a new mympdos-base-0.4.1 package with the bugfix.

jcorporation, 2021-01-06:
mympdos-base-0.4.1 is now online.

closed 2021-01-06 by jcorporation