jcorporation/myMPDos #8: Raspberry Pi Zero 2 W support

jnorth, 2022-02-27:
Hi, this project looks amazing, but I'm having some trouble getting it running on my Zero 2 W. This is my first Raspberry Pi project, so I could very well be missing some obvious knowledge.

After flashing my SD with myMPDos 0.9.4, the first issue I ran into was the ACT LED blinking 7 times. This looks to be a ["kernel.img not found"](https://raspberrypi.stackexchange.com/questions/41450/raspberry-pi-not-working/41455#41455) error. I noticed the kernel paths being set in the `config.txt` file, and based on [the docs](https://www.raspberrypi.com/documentation/computers/config_txt.html#model-filters) for that file I added the following:

```
[pi02]
kernel=boot/vmlinuz-rpi
initramfs boot/initramfs-rpi
```

This allowed the system to boot.

The second issue I ran into was the boot sequence giving the message: `No plan device found, falling back to eth0`. This led to errors when starting the `networking` init script, when `Checking network connectivity`, and finally having `mympdos-bootstrap.start` fail.

From my searching, it looks like at least some issues with the pi02w [were solved in Alpine 3.9.0](https://gitlab.alpinelinux.org/alpine/aports/-/issues/6959), but it looks like this project currently uses 3.15.

I'm still searching for other information. The most promising has been [this repo](https://github.com/lehmanjo/doc-alpine-linux-raspberry-pi-zero-w/blob/master/preparing.md). I initially thought it might be a missing dtoverlay to enable wifi on the pi02w, but I don't see anything relevant in [the overlay docs](https://github.com/raspberrypi/firmware/blob/master/boot/overlays/README).

I'll follow-up here if I find any other useful information.

jcorporation, 2022-02-28:
Thanks for testing myMPDos. I do not own a Zero 2 W, therefore I can not help a lot.

Changing config.txt is not ideal. It is overwritten on each update, but adding the lines to usercfg.txt should also work.

Searching for the wlan issue, I found this: https://gitlab.alpinelinux.org/alpine/aports/-/issues/13302

jcorporation, 2022-03-07:
Are you able to get the wlan running with above link?

jcorporation, 2022-03-22:
It seems Alpine Linux 3.15.1 fixes the firmware issue. I uploaded a new myMPDos 0.9.5 image.

Can you test and post feedback?

jnorth, 2022-04-16:
Thanks! The new image boots without the extra kernel paths now.

WiFi isn't working out of the box yet, still showing the `No wlan device found, falling back to eth0` message and `networking` errors.

To be honest, I'm not sure I know how to follow the instructions in the linked bug report. I've downloaded the referenced `brcmfmac43436-stio.raspberrypi,model-zero-2-w.*` files, but am unsure where to place them.

If I put them in the root of the myMPDos img, I do have them available in `/media/...`, but if I try copying them into the `/lib/firmware/brcm` directory as directed when booted I find out that it's a read-only file system. That makes sense—it's clear that a read-only mount is intended for this project, but I'm unsure if that means I have to build the myMPDos image myself with the files, or if there is a good way to do it with the official release image.

jcorporation, 2022-05-30:
I think this comment points out your problem: https://gitlab.alpinelinux.org/alpine/aports/-/issues/13302#note_223967

jcorporation, 2022-05-31:
Fix is in alpine edge and waits for testers.

jcorporation, 2022-08-13:
https://gitlab.alpinelinux.org/alpine/aports/-/commit/106fdd11ff134d5bb994c10e0f0e0dfc89cb9357

This commit adds the required firmware package to the rpi build script. I hope next alpine version 3.16.1 fixes this issue finally.

jcorporation, 2022-10-16:
Final fix must be merged:
https://gitlab.alpinelinux.org/alpine/alpine-conf/-/merge_requests/108

jcorporation, 2022-11-29:
I am unsure, but the fix could be in Alpine Linux 3.17.0. A myMPDos update is already available. Could you check?

pirateradiohack, 2022-12-11:
Hi. I also find this project amazing and am willing to try to incorporate it into my project. It's a huge time saver. I also happen to use a Raspberry Pi Zero W 2. 

I have just burnt the SD card with myMPDos v1.1.0 following the instructions and booted the device with it.  
At the moment I can't plug a screen so the only feedback I get are the green LED and the router leases.  
The green LED stays green, without flickering at all.  
And the router does not deliver any new lease (I've been refreshing the list for the past 10 / 15 minutes). 

So I guess something went wrong. I will try to find a way to plug a screen soon to get morefeed back and report back.

jcorporation, 2022-12-12:
>  I also find this project amazing and am willing to try to incorporate it into my project. 

That's great! I am glad to support you. Your transistor project looks very cool ;)

pirateradiohack, 2022-12-17:
Thank you very much @jcorporation I'm trying :)

My hope in trying a project based on Alpine is that the boot process would be much snappier. We'll see.

Regarding the current issue, I have plugged a screen and can report the following error appears at boot:

```
ERROR: musl-1.2.3-r4: BAD archive
ERROR: busybox-1.35.0r29.post-install: script exited with error 127
ERROR: alpine-baselayout-3.4.0-rc0.pre-install: script exited with error 127
ERROR: alpine-baselayout-3.4.0-rc0.post-install: script exited with error 127
ERROR: openrc-0.45.2-r6.post-install: script exited with error 127
ERROR: busybox-1.35.0-r29.trigger: script exited with error 127
/sbin/init not found in new root. Launching emergency recovery shell
Type exit to continue boot.
sh: can't access tty: job control turned off
/ #
```

I don't know if this is the same problem as the one described in this issue. I can move it to its own issue if required. 

I have no idea what could be the cause of that. If you have seen this before I'll gladly take your advice.
Meanwhile, the `BAD archive` part calls my attention and I think I'll simply try to burn the image again from scratch. 

Will report back later.

pirateradiohack, 2022-12-17:
I can confirm that re-flashing the card was sufficient to solve this particular issue.
And boot time does not feel considerably faster than with Raspbian.

jcorporation, 2022-12-17:
Alpine does a fresh package install in ram on each boot and then loads the overlay from sd. The benefit is a clean system after each boot and no problems with the sd-card after turning it off hard.

jcorporation, 2022-12-17:
> Hi. I also find this project amazing and am willing to try to incorporate it into my project. It's a huge time saver. I also happen to use a Raspberry Pi Zero W 2.

You you run myMPDos successfully on a PI Zero W2?

pirateradiohack, 2022-12-17:
Sorry I completely forgot to mention: yes, it works perfectly on the `Model: Raspberry Pi Zero 2 W Rev 1.0`

jcorporation, 2022-12-17:
@jnorth : can you confirm it too?

Then I can close this issue and add this model to known working devices list.

jnorth, 2022-12-18:
Boom. The kernel and WiFi are working out of the box now! Thank you so much for tracking this and getting a new release out!

jcorporation, 2022-12-19:
Great!

closed 2022-12-19 by jcorporation