jcorporation/myMPDos #9: Doesn't Work on Raspberry Pi 2B v1.2 (aarch64)

acocce, 2022-03-22:
Hi,
I installed myMPDos v0.9.4 to Raspberry Pi 2 Model B V1.2 (aarch64), but it didn't work. How do I adapt myMPDos to run on the hardware? I tried the following so far.

I added the following lines to the usercfg.txt
```
[pi2]
kernel=boot/vmlinuz-rpi
initramfs boot/initramfs-rpi
```

After booting the Raspberry Pi, I confirmed a specific IP address was leased successfully.

Then, I followed the [After first start](https://github.com/jcorporation/myMPDos/wiki/After-first-start) procedure, but failed to access the myMPD via web interface. The web browser (Firefox) displayed the following message.
> Unable to connect
> 
> Firefox can’t establish a connection to the server at 192.168.1.6.
> 
> The site could be temporarily unavailable or too busy. Try again in a few moments.
> If you are unable to load any pages, check your computer’s network connection.
> If your computer or network is protected by a firewall or proxy, make sure that Firefox is permitted to access the Web.

I also tried SSH access then the following message was displayed.
> Welcome to Alpine!
> 
> The Alpine Wiki contains a large amount of how-to guides and general
> information about administrating Alpine systems.
> See <http://wiki.alpinelinux.org/>.
> 
> You can setup the system with the command: setup-alpine
> 
> You may change this message by editing /etc/motd.
> 
> myMPDos:~#

Since I'm not so familiar with the CUI, I gave up going further. Any help is highly appreciated.

jcorporation, 2022-03-22:
It seems the bootstrap script was not running. Please reimage your sdcard and make your usercfg changes before booting the first time and post any errors occurring at startup. I own no pi 2b, so I can not test here.

acocce, 2022-03-23:
I installed the v0.9.5 image then confirmed everything worked successfully.
Thank you so much for your prompt support.
Cheers!

closed 2022-03-23 by acocce