jcorporation/myMPDos #11: I am unable to control audio volume

pirateradiohack, 2022-12-17:
myMPD shows `Volume: Volumecontrol disabled`.

I have switched the audio hat thanks to `audiohat.sh` to `hifiberry-dac` and the sound works but seems to be turned to maximum volume. 

It seems to be at the OS level since `alsamixer` shows `This sound device does not have any controls.` 
I am pretty sure I can control the sound device under Raspbian. Which makes me think it should work as well with Alpine. 

I am going to investigate but in the meantime if you have any experience or suggestion regarding setting the audio volume I'll gladly take it.

Thanks.

jcorporation, 2022-12-17:
Eventually Raspian enables the software mixer of mpd by default. myMPDos disables it for sound quality reasons.

pirateradiohack, 2022-12-17:
Do I understand correctly that it leaves us with only hardware volume controlled DACs for myMPDos?

jcorporation, 2022-12-17:
You can edit `/etc/mympdos/mympdos.conf` and change the `SOFTWARE_MIXER_FALLBACK` to `true`.

Run `/usr/bin/configmpd.sh`aftwards and restart mpd. This option is also available in the bootstrap file: https://github.com/jcorporation/myMPDos/blob/master/mympdos/bootstrap-advanced.txt

pirateradiohack, 2022-12-17:
ho great! Thanks for providing a fallback option!

pirateradiohack, 2022-12-17:
If I remember correctly there is also the possibility to control the DAC with alsa and then tell MPD to use the alsa master as its volume control. Might get better results in terms of sound quality.

closed 2022-12-17 by pirateradiohack

jcorporation, 2022-12-17:
It could be an option for the HifiBerry overlay? The default MPD configuration uses the alsa device directly and does not use the alsa mixer.

pirateradiohack, 2022-12-17:
> The default MPD configuration uses the alsa device directly

That sounds good to me.

jcorporation, 2022-12-17:
https://jcorporation.github.io/myMPDos/advanced-topics/customize-mpd