jcorporation/myMPDos #12: I find it inconvenient to have to switch to write mode each time I want to favorite a web radio in myMPD.

pirateradiohack, 2022-12-17:
I understand and support the point of having the OS in memory.

But I also find it inconvenient to have to remember that and toggle the read / write mode each time I want to save a web radio as a favorite in myMPD. Or maybe I misunderstand how it works, but if I don't toggle the write mode and save a favorite then it appears to be saved but when I reboot it's not there any more. Toggling the write mode does indeed save the favorite upon reboot. 

Do you have any recommendation? Ideally leaving the OS in read only in memory and having a permanent write mode just to save the favorites would be great.
Or maybe a myMPDos configuration that allows to leave the OS in write mode all the time (also defeating the purpose of having the OS in memory if I understand correctly).

jcorporation, 2022-12-17:
You should use the savereboot.sh script. It dumps the mympd state to disc, issues `lbu commit` before reboot. Alpine saves the state in an overlay file and applies it on startup. This is described in the Alpine documentation.

pirateradiohack, 2022-12-18:
Thanks for pointing in that direction. Will check this now.

pirateradiohack, 2022-12-18:
ok I get how it works. Sounds good to me.

How do you recommend using it?
I mainly see two ways I would use the shutdown / reboot in my project:

- from the myMPD web interface: do the `reboot` and `shutdown` menu options use those scripts?
- from the GPIO buttons: I will then use those scripts myself when programming the GPIOs. 

Another doubt: what would be the difference in using those scripts and (de)activating the read / write state with the `RW-Library` option menu in myMPD?

I am mainly thinking about the case of a hard shutdown here.
Do I understand it correctly if I say that the `save{reboot|shutdown}.sh` scripts won't be executed in case of a hard shutdown and thus the newly added content will be lost, while using the `RW-Library` allows to write directly to the FS in real time and thus a hard shutdown would still retain the newly written data?

jcorporation, 2022-12-18:
You are mixing to different functions:

1. `save{reboot|shutdown}.sh scripts` are there to save the state and reboot / shutdown, there is also an `save.sh` script to only save the state.
2. The `RW-Library` function toggles the data partitions rw/ro mode to allow users to copy music files in the library directory.

This are two completely different functions.

If you hard shutdown the device the state is lost anyway.

> from the myMPD web interface: do the reboot and shutdown menu options use those scripts?

yes

> from the GPIO buttons: I will then use those scripts myself when programming the GPIOs.

https://jcorporation.github.io/myMPDos/advanced-topics/power-on-and-off-with-gpio

pirateradiohack, 2022-12-18:
Ok thanks for clarifying. 

Indeed my webradio favorites do not disappear when I reboot from the web interface. That is great. 

Regarding `mygpiod` I think it's great to have it available from within the project. But for more "advanced" uses (attribute various functions to the same button depending on the length of the push, handling knobs...) I still like to use [gpiozero](https://gpiozero.readthedocs.io/en/stable/).

closed 2022-12-18 by pirateradiohack