jcorporation/myMPDos #13: Build fails at stage 3, can't open image file

pirateradiohack, 2022-12-28:
I think I have followed the steps and installed the dependencies (this is on Arch Linux), but yet when trying to build the image I get the following  output:

```
./build.sh build                                                                                                                                                                                                                                         
Building for aarch64                                                                                                                                                                                                                                        
Checking dependencies                                                                                                                                                                                                                                       
myMPDos build stage 1: Download                                                                                                                                                                                                                             
Getting alpine-netboot-3.17.0-aarch64.tar.gz                                                                                                                                                                                                                
Getting alpine-rpi-3.17.0-aarch64.tar.gz                                                                                                                                                                                                                    
myMPDos build stage 2: Create build image                                                                                                                                                                                                                   
8192+0 records in                                                                                                                                                                                                                                           
8192+0 records out                                                                                                                                                                                                                                          
8589934592 bytes (8.6 GB, 8.0 GiB) copied, 70.0024 s, 123 MB/s                                                                                                                                                                                              
Checking that no-one is using this disk right now ... OK                                                                                                                                                                                                    
                                                                                                                                                                                                                                                            
Disk myMPDos-build.img: 8 GiB, 8589934592 bytes, 16777216 sectors                                                                                                                                                                                           
Units: sectors of 1 * 512 = 512 bytes                                                                                                                                                                                                                       
Sector size (logical/physical): 512 bytes / 512 bytes                                                                                                                                                                                                       
I/O size (minimum/optimal): 512 bytes / 512 bytes                                                                                                                                                                                                           
                                                                                                                                                                                                                                                            
>>> Created a new DOS disklabel with disk identifier 0x48256963.                                                                                                                                                                                            
myMPDos-build.img1: Created a new partition 1 of type 'W95 FAT32' and of size 1024 MiB.                                                                                                                                                                     
myMPDos-build.img2: Done.                                                                                                                                                                                                                                   
                                                                                                                                                                                                                                                            
New situation:                                                                                                                                                                                                                                              
Disklabel type: dos                                                                                                                                                                                                                                         
Disk identifier: 0x48256963                                                                                                                                                                                                                                 
                                                                                                                                                                                                                                                            
Device             Boot Start     End Sectors  Size Id Type                                                                                                                                                                                                 
myMPDos-build.img1 *        1 2097151 2097151 1024M  b W95 FAT32                                                                                                                                                                                            

The partition table has been altered.                          
Syncing disks.                                                 
Checking that no-one is using this disk right now ... OK       

Disk myMPDos-build.img: 8 GiB, 8589934592 bytes, 16777216 sectors                                                             
Units: sectors of 1 * 512 = 512 bytes                          
Sector size (logical/physical): 512 bytes / 512 bytes          
I/O size (minimum/optimal): 512 bytes / 512 bytes              
Disklabel type: dos                                            
Disk identifier: 0x48256963                                    

Old situation:                                                 

Device             Boot Start     End Sectors  Size Id Type    
myMPDos-build.img1 *        1 2097151 2097151 1024M  b W95 FAT32                                                              

myMPDos-build.img2: Created a new partition 2 of type 'Linux' and of size 7 GiB.                                              
myMPDos-build.img3: Done.                                      

New situation:                                                 
Disklabel type: dos                                            
Disk identifier: 0x48256963                                    

Device             Boot   Start      End  Sectors  Size Id Type                                                               
[... cut ...]
qemu-system-aarch64: -drive file=myMPDos-build.img,format=raw: Could not open 'myMPDos-build.img': No such file or directory  
myMPDos build stage 4: Saving packages                         
losetup: myMPDos-build.img: failed to set up loop device: No such file or directory                                           
mount: mnt: mount point does not exist.                        
       dmesg(1) may have more information after failed mount system call.
```

Do you have any hint on what I should be looking for? Thanks.

jcorporation, 2022-12-28:
Try starting stage 3 manually: `./build.sh 3`

pirateradiohack, 2022-12-28:
Indeed, running the missing stages manually did work and created an image. Thanks.

Specifically I did:
`./build.sh 3`
`./build.sh 4`
`./build.sh 5`

Is it just happening to me or is it a common bug you are planning to fix?

jcorporation, 2022-12-28:
This will be fixed.

jcorporation, 2022-12-28:
Should be fixed with 4a0fe81.

pirateradiohack, 2022-12-29:
Thanks for the quick update. 

I have tried with https://github.com/jcorporation/myMPDos/commit/4a0fe814307aa0962e1a82dea911fb9ea0f69bf5 but in my case it did not produce an image. It did not return any error message though.

The `tmp` directory has been created, with a `aarch64` directory within it. But no sign of the `images` directory.

I also tried to run the different stages independently but to no avail. 

I'm not sure what went wrong. Should I try it in a different way?

jcorporation, 2022-12-29:
No error and no other output? Let me test it again locally. You could run it with `bash -x ./build.sh b` to get debug output.

pirateradiohack, 2022-12-29:
OK. I have launched the build with `bash -x ./build.sh b `. Let's see where it gets us.

By the way, I know it's a bit off-topic, but I don't know where else to ask. I am trying to build the image a bit out of curiosity, but also because I would like to produce the same image but pre-loaded with a few Python modules (along with their dependencies) that I need to run my project. Do you think building the image with some modifications is the best way (and what are your recommendations, if any as to how best to modify it?). Or do you recommend a completely different approach (like modifying a pre-built image somehow?)

pirateradiohack, 2022-12-29:
Meanwhile, the build has finished successfully:

```
+ echo 'Image myMPDos-aarch64-1.1.0-20221229.img created successfully'
Image myMPDos-aarch64-1.1.0-20221229.img created successfully
+ '[' false = true ']'
+ echo ''

+ echo 'Next step is to burn the image to a sd-card and'
Next step is to burn the image to a sd-card and
+ echo 'create the bootstrap.txt file.'
create the bootstrap.txt file.
+ echo 'There are samples in the image.'
There are samples in the image.
+ echo ''

+ exit 0

```

But yet no `images` directory and no sign of the file in my whole `home`.

I will try to find a sign in the output.

jcorporation, 2022-12-29:
> By the way, I know it's a bit off-topic, but I don't know where else to ask.

Simply open a new issue.

You can install extra software at bootstrap time with the the `EXTRA_SOFTWARE` config option.

> But yet no images directory and no sign of the file in my whole home.

I tested it again and it works on my local setup. The folder `images` should appear in the same folder as the `build.sh` script resides.

pirateradiohack, 2022-12-29:
I have completely removed the myMPDos folder from my FS and cloned it again, just in case. Now when I run the build there is a `images` directory created with the image inside. Great! Seems to work.

jcorporation, 2022-12-29:
Great, I saw that the documentation on building was a bit outdated. I will update it.

closed 2022-12-29 by jcorporation

pirateradiohack, 2022-12-29:
Thanks for your efforts. This project is awesome!