jcorporation/myMPDos #15: SD card does not auto mount like a USB stick does

jnorth, 2023-01-01:
I don't believe this is an issue with myMPDos specifically, but wanted to raise here in case it was a configuration value set by the image. Feel free to close if this is out of this project's scope.

When I plug a USB stick into the Raspberry Pi, it's files are automatically mounted and accessible.

When I plug in a USB card reader, its device is detected, but when I insert an SD card into that it is not mounted or accessible.

Lastly, if I plug in the SD card before inserting the USB card reader, it is automatically mounted and accessible. Strange!

[RaspberryPi] -- Micro USB -- [USB Card Reader] -- SD Slot -- [SD Card]

I've tried with a couple different USB card readers, and a couple different SD cards, and non seem to support hot swapping the SD card. I've posted a question to the unix/linux Stack Exchange, and hopefully someone can point me in the right direction. More details in that post if this seems like something that could be configured through this distribution:

https://unix.stackexchange.com/questions/730184/with-a-usb-sd-card-reader-the-sd-card-only-shows-up-if-it-is-inserted-before-th

jcorporation, 2023-01-01:
myMPDos uses a custom script for automounting: `/usr/bin/automount.sh`. This script is triggered by mdev. Config file is `/etc/mdev.conf`. You should look, if mdev triggers an event on sd-card insertion.

jnorth, 2023-01-01:
Thanks, that helps point me in the right direction.

mdev is indeed not triggering events for the sd-card insertion/removal, so the issue is happening in mdev or somewhere further upstream. The mdev config looks good, as does the auto mount.sh script 👍.

I'll update here if I find a solution in case it helps anyone else, but I'll close the ticket since it's not at the myMPDos level.

closed 2023-01-01 by jnorth

jnorth, 2023-04-03:
Just wanted to share an update here in case it helped anyone else.

There very well could be a better solution which enables `mdev` to trigger events for the SD card being inserted or removed. It seems like it could be a missing configuration or module somewhere in the system, but nothing I did was able to get an event to fire that I could use.

Instead, I installed `udev` using the `setup-devd` command, and modified the udev rules to replicate some of what myMPDos is doing with the `mdev.conf` and `automount.sh` scripts. It works!

NOTE: This still feels pretty hacky. When inserting the SD card, I get an `add` event and mount the device. Removing the card however does not fire a `remove` event if the music files are in use. Instead, I also listen for `change` events, and unmount on those if I detect the change event firing for a previously mounted device. Good enough for me for now, but I think the right approach would be to tell MPD to remove the files on `change` and hopefully that would let the `remove` event trigger normally.

Anyway, here is what I've done:

### Replace mdev with udev

```
setup-devd
```

Enter `udev` and choose `y` to repopulate hardware.

### Add udev rules file to listen for sd devices

Create a file `/etc/udev/rules.d/10-sd.rules`:

```
KERNEL=="sd[a-z]*", RUN+="/usr/bin/automount.sh"
```

### Modify the automount.sh script

Replace `/usr/bin/automount.sh` with:

```
#!/bin/sh
#
# SPDX-License-Identifier: GPL-3.0-or-later
# myMPDos (c) 2020-2022 Juergen Mang <mail@jcgames.de>
# https://github.com/jcorporation/myMPDos

MDEV="$(basename $DEVNAME)"
MOUNTDIR="/mnt/$MDEV"

if [ "$ACTION" = "add" ]
then
  # Mount device
  mkdir "$MOUNTDIR"
  mount -oro,noatime "$DEVNAME" "$MOUNTDIR"

  # Update music database
  [ -f /run/mpd/pid ] && mpc update

elif [ "$ACTION" = "change" ]
then
  # Scan each mount point
  # The change action happens at the root (sda vs sda1), so we loop through and unmount any partitions using that root
  for MOUNT in /mnt/*/
  do
    MOUNTNAME=$(basename $MOUNT)

    if [ "$(echo $MOUNTNAME | sed 's/[0-9]//g')" = "$MDEV" ]
    then
      # Unmount device
      umount "$MOUNT"
      rmdir "$MOUNT"

      # Update music database
      [ -f /run/mpd/pid ] && mpc update
    fi
  done

elif [ "$ACTION" = "remove" ]
then
  # Unmount device
  umount "$MOUNTDIR"
  rmdir "$MOUNTDIR"

  # Update music database
  [ -f /run/mpd/pid ] && mpc update
fi
```

You may need to reload the udev rules to get it to follow the new `10-sd.rules` file.

```
udevadm control --reload-rules && udevadm trigger
```

jcorporation, 2023-04-03:
Thanks for sharing this!

mdev has a change event. Had you tried this event?