jcorporation/myMPDos #17: Self built image cannot use myMPDos repositories

pirateradiohack, 2023-01-06:
I have successfully built the image but the first boot fails at:

```
adding repositories and upgrade
...
Adding myMPDos archiv signing public key
...
fetch https://raw.githubusercontent.com/jcorporation/myMPDos/master/repository/aarch64/APKINDEX.tar.gz
ERROR: https://raw.githubusercontent.com/jcorporation/myMPDos/master/repository: UNTRUSTED signature
WARNING: Ignoring https://raw.githubusercontent.com/jcorporation/myMPDos/master/repository: No such file or directory

```

the above error repeats another 2 times. Then the base Alpine linux installs correctly but there is no MPD running nor web server.

I have built the image with `./build.sh build`.

Did I miss something?

jcorporation, 2023-01-06:
That is a bug. The build script should always include my archive signing key too.

pirateradiohack, 2023-01-07:
ok and is there anything I can do? Is this something you need to fix on your side?

jcorporation, 2023-01-07:
I must fix the build / bootstrap fix.

jcorporation, 2023-01-08:
Should be fixed in current master, please test.

pirateradiohack, 2023-01-08:
I'm happy to report the error has disappeared, your repository is correctly added and the final result works as expected.

Thank you.

closed 2023-01-08 by pirateradiohack