jcorporation/myMPDos #18: rpi-dac / i2s-dac do not show up in the results when running audiohat.sh.

rhoering, 2023-01-28:
rpi-dac / i2s-dac do not show up in the results when running audiohat.sh. They are present in the overlays/README.

myMPDos:/usr/bin# ./audiohat.sh
Downloading audio hats description
  1) allo-boss-dac-pcm512x-audio
  2) allo-boss2-dac-audio
  3) allo-digione
  4) allo-katana-dac-audio
  5) allo-piano-dac-pcm512x-audio
  6) allo-piano-dac-plus-pcm512x-audio
  7) applepi-dac
  8) chipdip-dac
  9) hifiberry-amp
  10) hifiberry-amp100
  11) hifiberry-amp3
  12) hifiberry-dac
  13) hifiberry-dacplus
  14) hifiberry-dacplusadc
  15) hifiberry-dacplusadcpro
  16) hifiberry-dacplusdsp
  17) hifiberry-dacplushd
  18) hifiberry-digi
  19) hifiberry-digi-pro
  20) iqaudio-codec
  21) iqaudio-dac
  22) iqaudio-dacplus
  23) iqaudio-digi-wm8804-audio
  24) justboom-digi
  25) merus-amp
  26) pibell

rhoering, 2023-01-28:
Introduction
============

This directory contains Device Tree overlays. Device Tree makes it possible
to support many hardware configurations with a single kernel and without the
need to explicitly load or blacklist kernel modules. Note that this isn't a
"pure" Device Tree configuration (c.f. MACH_BCM2835) - some on-board devices
are still configured by the board support code, but the intention is to
eventually reach that goal.

On Raspberry Pi, Device Tree usage is controlled from /boot/config.txt. By
default, the Raspberry Pi kernel boots with device tree enabled. You can
completely disable DT usage (for now) by adding:

    device_tree=

to your config.txt, which should cause your Pi to revert to the old way of
doing things after a reboot.

In /boot you will find a .dtb for each base platform. This describes the
hardware that is part of the Raspberry Pi board. The loader (start.elf and its
siblings) selects the .dtb file appropriate for the platform by name, and reads
it into memory. At this point, all of the optional interfaces (i2c, i2s, spi)
are disabled, but they can be enabled using Device Tree parameters:

    dtparam=i2c=on,i2s=on,spi=on

However, this shouldn't be necessary in many use cases because loading an
overlay that requires one of those interfaces will cause it to be enabled
automatically, and it is advisable to only enable interfaces if they are
needed.

Configuring additional, optional hardware is done using Device Tree overlays
(see below).

GPIO numbering uses the hardware pin numbering scheme (aka BCM scheme) and
not the physical pin numbers.

raspi-config
============

The Advanced Options section of the raspi-config utility can enable and disable
Device Tree use, as well as toggling the I2C and SPI interfaces. Note that it
is possible to both enable an interface and blacklist the driver, if for some
reason you should want to defer the loading.

Modules
=======

As well as describing the hardware, Device Tree also gives enough information
to allow suitable driver modules to be located and loaded, with the corollary
that unneeded modules are not loaded. As a result it should be possible to
remove lines from /etc/modules, and /etc/modprobe.d/raspi-blacklist.conf can
have its contents deleted (or commented out).

Using Overlays
==============

Overlays are loaded using the "dtoverlay" config.txt setting. As an example,
consider I2C Real Time Clock drivers. In the pre-DT world these would be loaded
by writing a magic string comprising a device identifier and an I2C address to
a special file in /sys/class/i2c-adapter, having first loaded the driver for
the I2C interface and the RTC device - something like this:

    modprobe i2c-bcm2835
    modprobe rtc-ds1307
    echo ds1307 0x68 > /sys/class/i2c-adapter/i2c-1/new_device

With DT enabled, this becomes a line in config.txt:

[... 3971 lines cut ...]
        backlight-gpio          Defines a GPIO to be used for backlight control
                                (default of none).
        backlight-pwm           Defines a PWM channel to be used for backlight
                                control (default of none). NB Disables audio
                                headphone output as that also uses PWM.
        backlight-pwm-chan      Choose channel on &pwm node for backlight
                                control.
                                (default 0).
        backlight-pwm-gpio      GPIO pin to be used for the PWM backlight. See
                                pwm-2chan for valid options.
                                (default 18 - note this can only work with
                                 rgb666-padhi).
        backlight-pwm-func      Pin function of GPIO used for the PWM
                                backlight.
                                See pwm-2chan for valid options.
                                (default 2).
        backlight-def-brightness
                                Set the default brightness. Normal range 1-16.
                                (default 16).
        rotate                  Display rotation {0,90,180,270} (default 0)

Name:   vc4-kms-dpi-hyperpixel2r
Info:   Enable the KMS drivers for the Pimoroni HyperPixel2 Round DPI display.
        Requires vc4-kms-v3d to be loaded.
Load:   dtoverlay=vc4-kms-dpi-hy

jcorporation, 2023-01-29:
The script filters only for `audio card` in the readme, bug fix is to filter for `(audio card|soundcard|sound card)`.

jcorporation, 2023-01-29:
Fixed with mympdos-base 0.7.5.

closed 2023-01-29 by jcorporation