jcorporation/myMPDos #21: mympdos 1.2.2 mongoose cant bind port

jerta, 2023-07-26:
I launch a fresh 1.2.2, mympd doesn't start, so i try :
myMPDos:~# mympd
12:11:53 NOTICE   mympd     Cache dir: "/var/cache/mympd"
12:11:53 NOTICE   mympd     Work dir: "/var/lib/mympd"
12:11:53 NOTICE   mympd     Config dir: "/var/lib/mympd/config"
12:11:53 NOTICE   mympd     Setting loglevel to NOTICE
12:11:53 NOTICE   mympd     Starting myMPD 11.0.1
598a7  1 mongoose.c:4685:mg_open_listen bind: 98
598a7  1 mongoose.c:3713:mg_listen      Failed: http://[::]:80, errno 98
12:11:53 EMERG    mympd     Can't bind to http://[::]:80
Exiting erroneous, thank you for using myMPD

jcorporation, 2023-07-27:
The newest myMPD binds as default to [::], but myMPDos disables IPv6 by default and set's bindv6only to 1.

Simplest fix should be:

```sh
echo "0.0.0.0" > /var/lib/mympd/config/http_host
savereboot.sh
```

Next patch version of myMPD will enable IPv6 by default. Thanks for submitting this bug.

bTronics, 2023-07-28:
Hello, 
I made also a fresh install of the latest myMPDos  and have the same issue. I did:

echo "0.0.0.0" > /var/lib/mympd/config/http_host
savereboot.sh

But I have still no connection (still the same error)

jcorporation, 2023-07-28:
What is the value of this setting after reboot?

bTronics, 2023-07-28:
The value was 0.0.0.0 , I think the problem was the IP_TYPE="dhcp". I uncomment this and made again a fresh install and now I have connection. 
I have just one little problem with the startscreen in mympd. when I add an icon, the next time I start up (after a save power off / save reboot) the start screen returns empty.

jcorporation, 2023-07-28:
myMPDos runs completely in ram. You must use the savereboot.sh or saveshutdown.sh or the scripts in the myMPD menu. This should dump the states and issues a lbu_commit.

closed 2023-07-29 by jcorporation