jcorporation/myMPDos #34: Best way to use gpiozero

pirateradiohack, 2023-11-24:
Hello,

I use some additional hardware to control the volume, next / previous, on / off actions.
See

https://shop.pimoroni.com/products/onoff-shim?variant=41102600138
https://shop.pimoroni.com/products/raspio-analog-zero

I am comfortable controlling those with gpiozero : https://gpiozero.readthedocs.io/en/latest/

What would be the best way to install gpiozero on myMPDos ?

All I can do is installing some alpine packages by listing them in the bootstrap.txt, but  gpiozero recommends installing through pip (the python package manager).

I tried installing pip on a running myMPDos but it did not work. I guess because of the in ram system. The error is no space left on the device, which makes sense.
So, before fiddling more in the wrong direction, is there a recommended way ?

Thanks

jcorporation, 2023-11-24:
I can not recommend any way. The maximum size of the root filesystem is limited by the amount of ram your raspberry have.  Because of this limitation: install only the minimum of required packages or install the packages into a subdir of `/media/mmcblk0p2`.

myMPDos integrates myGPIOd that uses libgpiod. I use it to run script on button press. The functionality is at the moment rather limited, but it could be extended.

pirateradiohack, 2023-11-25:
Thanks.

I have considered myGPIOd. But I would need more features. Like long presses and volume knob.

I think in the meantime I will try to install myMPD on a classic RaspberryPi OS which comes bundled with gpiozero.

jcorporation, 2023-11-25:
This should work without problems. I would recommend DietPi, it ships myMPD. 

If you like, you can open issues with features for myGPIOd in the myGPIOd repository.

closed 2023-11-25 by jcorporation