jcorporation/myMPDos #35: mygpiod user and gpio group appears to be missing in myMPDos 1.3.0

gd-99, 2023-12-07:
I installed the current myMPDos version 1.3.0 image to a Raspberry Pi 3B+. Install appeared to go without error. I added a push button switch to gpio pins 5 & 6, and followed the set-up documentation for myMPD. When trying to run the service, it failed with an error to tell me, in effect, the mygpiod user and gpio group were missing.

I manually added the user and group after which the the service ran correctly. For some reason I had to reboot for the doas configuration to be read. Before rebooting I would get a "not permitted" error when I pushed the button. Although a long time Linux user this is my first exposure to Alpine Linux, and by RTFM, may have saved me a reboot or two. I did try reinstalling myGPIOd using apk, but this didn't recreate the user or group for me.

Otherwise the myMPDos set-up couldn't be easier and happy to say all working now. Very nice project.

jcorporation, 2023-12-07:
The user should be created by the pre-install script of myGPIOd. Are there any errors installing the mygpiod package?

gd-99, 2023-12-07:
At the time I didn't notice any errors in the console. As I say I am very new to Alpine. I can't see any errors in /var/logs - is this where apk would store any logs? At the time I reinstalled myGPIOd via apk, I had already started to try and resolve the error and had manually added the user and group - however they didn't get added to the shadow passwd files. Would this have stopped the pre-install script from overwiting my manual changes?

jcorporation, 2023-12-08:
The user should be added unconditionally: https://github.com/jcorporation/myGPIOd/blob/master/contrib/packaging/alpine/mygpiod.pre-install

jcorporation, 2023-12-13:
The libgpio package from the Alpine Linux repository does not create the gpio user anymore. Fix is easy. I push a new myGPIOd package today.

closed 2023-12-13 by jcorporation