jcorporation/myMPDos #38: Nothing runs on first boot

stratosgear, 2024-04-11:
Followed the instructions to the letter and although I can ssh to my pi, no mpd/mympd services are running at all (nothing available  at https://my.pi.ip)

Redid the installation (burned again a new sd card) but same result.

I noticed  that:
```
# myMPDos repository
MYMPDOS_REPOSITORY="https://raw.githubusercontent.com/jcorporation/myMPDos/master/repository"
```

resolves to a 404 page. https://raw.githubusercontent.com/jcorporation/myMPDos/master/repository

That doesn't seem right, does it?

stratosgear, 2024-04-11:
Hmmm, I guess that url is used as a root folder, based on which some packages are supposed to be looked up and installed?

My process list immediattely after booting the first time looks like:

```
pi3:~# pstree
init-+-getty
     |-ntpd
     |-sshd---sshd---ash---pstree
     |-syslogd
     `-udhcpc
```

Am I supposed to be installing something? The docs make it seem that everything should be done for me!

dmesg looks "clean":

```
pi3:~# dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 6.6.14-0-rpi (buildozer@build-3-19-aarch64) (gcc (Alpine 13.2.1_git20231014) 13.2.1 20231014, GNU ld (GNU Binutils) 2.41) #1-Alpine SMP PREEMPT Fri Jan 26 13:41:25 UTC 2024
[    0.000000] KASLR enabled
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x0000000037400000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x0000000037400000..0x000000003b3fffff (65536 KiB) map reusable linux,cma
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000] On node 0, zone DMA: 19456 pages in unavailable ranges
[    0.000000] percpu: Embedded 30 pages/cpu s85224 r8192 d29464 u122880
[    0.000000] pcpu-alloc: s85224 r8192 d29464 u122880 alloc=30*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] CPU features: detected: ARM erratum 843419
[    0.000000] CPU features: detected: ARM erratum 845719
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_headphones=0 snd_bcm2835.enable_headphones=0 snd_bcm2835.enable_hdmi=0 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  modules=loop,squashfs,sd-mod,usb-storage quiet console=tty1
[    0.000000] Unknown kernel command line parameters "modules=loop,squashfs,sd-mod,usb-storage", will be passed to user space.
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 238896
[    0.000000] mem auto-init: stack:all(zero), heap alloc:off, heap free:off
[    0.000000] software IO TLB: area num 4.
[    0.000000] software IO TLB: mapped [mem 0x0000000031f80000-0x0000000035f80000] (64MB)
[    0.000000] Memory: 788980K/970752K available (12544K kernel code, 2204K rwdata, 4240K rodata, 3456K init, 1073K bss, 116236K reserved, 65536K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000] 	Trampoline variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: bcm2836_arm_irqchip_handle_irq
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000001] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[... 251 lines cut ...]
[   10.075367] bcm2835-isp bcm2835-isp: Register capture node 2 with media controller
[   10.075375] bcm2835-isp bcm2835-isp: Register capture node 3 with media controller
[   10.088533] bcm2835-isp bcm2835-isp: Device node output[0] registered as /dev/video20
[   10.093112] bcm2835-isp bcm2835-isp: Device node capture[0] registered as /dev/video21
[   10.093651] bcm2835-isp bcm2835-isp: Device node capture[1] registered as /dev/video22
[   10.094147] bcm2835-isp bcm2835-isp: Device node stats[2] registered as /dev/video23
[   10.094181] bcm2835-isp bcm2835-isp: Register output node 0 with media controller
[   10.094196] bcm2835-isp bcm2835-isp: Register capture node 1 with media controller
[   10.094205] bcm2835-isp bcm2835-isp: Register capture node 2 with media controller
[   10.094213] bcm2835-isp bcm2835-isp: Register capture node 3 with media controller
[   10.094667] bcm2835-isp bcm2835-isp: Loaded V4L2 bcm2835-isp
[   10.108492] snd_bcm2835: module is from the staging directory, the quality is unknown, you have been warned.
[   11.200049] NET: Registered PF_PACKET protocol family
[   11.310324] NET: Registered PF_INET6 protocol family
[   11.311408] Segment Routing with IPv6
[   11.311439] In-situ OAM (IOAM) with IPv6
[   11.513099] EXT4-fs (mmcblk0p2): re-mounted 7a18cfd9-bc4e-4040-be5e-edaa18ed0467 ro. Quota mode: none.
[   12.272349] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[   12.276194] smsc95xx 1-1.1:1.0 eth0: Link is Up - 100Mbps/Full - flow control off
pi3:~# 
```

How  else can I debug this? 

Thanks, btw!!!

jcorporation, 2024-04-11:
The 404 for the repository url is correct, there is no index page and directory listing is disabled.

On first boot the bootstrap script should run and configure myMPDos accordingly to your created bootstrap.txt file. It removes itself and reboots afterwards. Are you seeing any errors in this stage?

stratosgear, 2024-04-11:
Hmm, the pi3 is headless.  I'll have to scramble to get a display to it to see what happens... :(

I found the `mympdos/overlay/etc/local.d/mympdos-bootstrap.start` file that is doing the bootstrapping, and tried to do some redirects to a log file, but I see that the bootstrap is doing some heavy duty sd card repartitioning so I have nowhere to output (second :( )

I guess I *really* have to scramble for an external monitor... :sniff:

jcorporation, 2024-04-11:
Logging the output of the bootstrap script is a good idea. I will look into this for the next release.

stratosgear, 2024-04-11:
Ok.  Still painful! Even with an attached monitor the output scrolls too fast to read it all.  Had to boot a couple of times (and every time burning again the SD card from scratch and editing bootstrap.txt) and STILL not able to catch the error. :(

Had to try again and take a video of the boot process.

The mistake was on a custom package `s3fs` that in alpine it seems it is named `s3fs-fuse`.  The bootstrapper fails with a:

```
ERROR: unable to select packages:
  s3fs (no such package)
    required by: world(s3fs)
```

User error, indeed, but the end effect was REALLY hard and inconvenient to debug.

I think a permanent log would be a GREAT addition!

Sorry to sound so negative, I am not.  This just took MUCH longer than I was expecting, and I was not prepared for that.

jcorporation, 2024-04-11:
Ok, you add an invalid package name to the EXTRA_SOFTWARE variable?

jcorporation, 2024-04-11:
https://github.com/jcorporation/myMPDos/commit/90f01f31b0ba063cfbef30efe77f4d0bacec97b7

stratosgear, 2024-04-12:
Hey thanks!

I cloned the repo, run the `build.sh` script from the devel branch and burned the ./images/myMPDos-aarch64-1.4.2-20240412.img to my sd card with: ` sudo dd bs=4M if=./images/myMPDos-aarch64-1.4.2-20240412.img of=/dev/sde status=progress oflag=sync`

I mounted the sdcard with sudo mount /dev/sde1 /mnt/sdcard and I copied a simplyfied bootstrap of:

```
# Simple bootstrap configuration file for myMPDos
# For all available options refer to bootstrap-advanced.txt

ROOT_PASSWORD="someLongPass"
TIMEZONE="Europe/Madrid"

# If WLAN_ENABLED is not set to true eth0 is configured
WLAN_ENABLE="false"
#WLAN_SSID="ssid of your wlan"
#WLAN_PSK="psk of the ssid"

AUDIOHAT="iqaudio-dacplus"
``` 

Booting with this sd card, I get to connect to the pi but` pstree` still gives:
```
myMPDos:~# pstree
init-+-getty
     |-ntpd
     |-sshd---sshd---ash---pstree
     |-syslogd
     `-udhcpc
```
and ofcourse nothing is available on https://<my.ip> (neither on http)

I poweroff the pi and mount the sd card on my desktop and I find the new bootrstap.log that looks "ok":

```
✗  cat bootstrap.log

myMPDos 1.4.2 bootstrap script

 *   WARNING: you are stopping a sysinit service
 *   Executing: /lib/rc/sh/openrc-run.sh /lib/rc/sh/openrc-run.sh /etc/init.d/modloop stop
 *   Unmounting /.modloop ... [ ok ]
Creating data partition on sd card

The number of cylinders for this disk is set to 111129.
There is nothing wrong with that, but this is larger than 1024,
and could in certain setups cause problems with:
1) software that runs at boot time (e.g., old versions of LILO)
2) booting and partitioning software from other OSs
   (e.g., DOS FDISK, OS/2 FDISK)

Command (m for help): Partition type
   p   primary partition (1-4)
   e   extended
Partition number (1-4): First sector (1048576-31116287, default 1048576): Using default value 1048576
Last sector or +size{,K,M,G,T} (1048576-31116287, default 31116287): Using default value 31116287

Command (m for help): The partition table has been altered.
Calling ioctl() to re-read partition table
Waiting for /dev/mmcblk0p1 found with FS
 *   Executing: /lib/rc/sh/openrc-run.sh /lib/rc/sh/openrc-run.sh /etc/init.d/modloop start
 *   Mounting modloop  ...
 *   Verifying modloop
 [ ok ]
Waiting for /dev/mmcblk0p2 found with NOFS
Formating data partition on sd card
(1/5) Installing libblkid (2.39.3-r0)
(2/5) Installing libcom_err (1.47.0-r5)
(3/5) Installing e2fsprogs-libs (1.47.0-r5)
(4/5) Installing libuuid (2.39.3-r0)
[... 432 lines cut ...]
var/lib/mympd/config/ssl_san
var/lib/mympd/config/custom_cert
var/lib/mympd/config/pin_hash
var/lib/mympd/config/acl
var/lib/mympd/config/scriptacl
var/lib/mympd/config/lualibs
var/lib/mympd/config/covercache_keep_days
var/lib/mympd/config/loglevel
var/lib/mympd/config/save_caches
var/lib/mympd/config/mympd_uri
var/lib/mympd/config/stickers
var/lib/mympd/config/stickers_pad_int
var/lib/mympd/config/album_mode
var/lib/mympd/config/album_group_tag
var/lib/mympd/ssl/
var/lib/mympd/ssl/ca.key
var/lib/mympd/ssl/ca.pem
var/lib/mympd/ssl/server.key
var/lib/mympd/ssl/server.pem
Created /media/mmcblk0p2/myMPDos.apkovl.tar.gz
Successfully saved apk overlay files
```
I had the same experience with the published v1.4.1 image, but I get no bootstrap.log from there.

Am I still missing something here?  Why are the services not up?  Is there something else to try/debug?

jcorporation, 2024-04-12:
If you start the image in qemu it does not load the overlay created by the bootstrap process. This is expected.

Your bootstrap log looks good. The startup log after the first reboot is interesting:

- Is the overlay found and loaded?

stratosgear, 2024-04-12:
Justin case it might be usefull. Here is a dmesg from the latest boot (actually first boot of the newly burned sd card):

```
myMPDos:~# dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 6.6.14-0-rpi (buildozer@build-3-19-aarch64) (gcc (Alpine 13.2.1_git20231014) 13.2.1 20231014, GNU ld (GNU Binutils) 2.41) #1-Alpine SMP PREEMPT Fri Jan 26 13:41:25 UTC 2024
[    0.000000] KASLR enabled
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x0000000037400000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x0000000037400000..0x000000003b3fffff (65536 KiB) map reusable linux,cma
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000] On node 0, zone DMA: 19456 pages in unavailable ranges
[    0.000000] percpu: Embedded 30 pages/cpu s85224 r8192 d29464 u122880
[    0.000000] pcpu-alloc: s85224 r8192 d29464 u122880 alloc=30*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] CPU features: detected: ARM erratum 843419
[    0.000000] CPU features: detected: ARM erratum 845719
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_headphones=0 snd_bcm2835.enable_headphones=0 snd_bcm2835.enable_hdmi=0 bcm2708_fb.fbwidth=1920 bcm2708_fb.fbheight=1200 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  modules=loop,squashfs,sd-mod,usb-storage quiet console=tty1
[    0.000000] Unknown kernel command line parameters "modules=loop,squashfs,sd-mod,usb-storage", will be passed to user space.
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 238896
[    0.000000] mem auto-init: stack:all(zero), heap alloc:off, heap free:off
[    0.000000] software IO TLB: area num 4.
[    0.000000] software IO TLB: mapped [mem 0x0000000031f80000-0x0000000035f80000] (64MB)
[    0.000000] Memory: 788980K/970752K available (12544K kernel code, 2204K rwdata, 4240K rodata, 3456K init, 1073K bss, 116236K reserved, 65536K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000] 	Trampoline variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: bcm2836_arm_irqchip_handle_irq
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000001] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000404] Console: colour dummy device 80x25
[    0.000419] printk: console [tty1] enabled
[    0.000528] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=76800)
[    0.000549] pid_max: default: 32768 minimum: 301
[    0.000839] LSM: initializing lsm=capability,integrity
[    0.001105] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.001137] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.002258] cgroup: Disabling memory control group subsystem
[    0.004284] RCU Tasks: Setting shift to 2 and lim to 1 rcu_task_cb_adjust=1.
[    0.004460] RCU Tasks Trace: Setting shift to 2 and lim to 1 rcu_task_cb_adjust=1.
[    0.004765] rcu: Hierarchical SRCU implementation.
[    0.004774] rcu: 	Max phase no-delay instances is 1000.
[    0.006849] EFI services will not be available.
[    0.007242] smp: Bringing up secondary CPUs ...
[    0.008048] Detected VIPT I-cache on CPU1
[... 238 lines cut ...]
[   10.519704] bcm2835-isp bcm2835-isp: Device node capture[0] registered as /dev/video14
[   10.520102] bcm2835-isp bcm2835-isp: Device node capture[1] registered as /dev/video15
[   10.520470] bcm2835-isp bcm2835-isp: Device node stats[2] registered as /dev/video16
[   10.520500] bcm2835-isp bcm2835-isp: Register output node 0 with media controller
[... cut ...]
[   10.563650] snd_bcm2835: module is from the staging directory, the quality is unknown, you have been warned.
[   11.686840] NET: Registered PF_PACKET protocol family
[   11.797407] NET: Registered PF_INET6 protocol family
[   11.798470] Segment Routing with IPv6
[   11.798499] In-situ OAM (IOAM) with IPv6
[   12.017108] EXT4-fs (mmcblk0p2): re-mounted 264d0b14-f834-47a6-8006-3596c185d265 ro. Quota mode: none.
[   12.848686] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[   12.853478] smsc95xx 1-1.1:1.0 eth0: Link is Up - 100Mbps/Full - flow control off
```

jcorporation, 2024-04-12:
dmesg output is not useful. What says apk info? List it myMPD and MPD packages?

stratosgear, 2024-04-12:
`/var/log/messages` contain boring info:

```
Jan 11 14:43:03 myMPDos syslog.info syslogd started: BusyBox v1.36.1
Jan 11 14:43:03 myMPDos daemon.info init: starting pid 2463, tty '': '/sbin/openrc default'
Jan 11 14:43:03 myMPDos daemon.warn openrc[2463]: WARNING: clock skew detected!
Jan 11 14:43:03 myMPDos auth.info sshd[2519]: Server listening on 0.0.0.0 port 22.
Jan 11 14:43:03 myMPDos auth.info sshd[2519]: Server listening on :: port 22.
Jan 11 14:43:04 myMPDos daemon.info init: starting pid 2550, tty '/dev/tty1': '/sbin/getty 38400 tty1'
Apr 12 20:17:03 myMPDos auth.info sshd[2598]: Accepted password for root from 192.168.3.94 port 47078 ssh2
```

But apk info does NOT list the required packages:

```
myMPDos:~# apk info
WARNING: opening from cache http://dl-cdn.alpinelinux.org/alpine/v3.19/main: No such file or directory
WARNING: opening from cache http://dl-cdn.alpinelinux.org/alpine/v3.19/community: No such file or directory
WARNING: opening from cache https://raw.githubusercontent.com/jcorporation/myMPDos/master/repository: No such file or directory
alpine-base
alpine-baselayout
alpine-baselayout-data
alpine-conf
alpine-keys
alpine-release
apk-tools
busybox
busybox-binsh
busybox-mdev-openrc
busybox-openrc
busybox-suid
ca-certificates-bundle
doas
e2fsprogs
e2fsprogs-libs
ifupdown-ng
libblkid
libc-utils
libcap2
libcom_err
libcrypto3
libedit
libncursesw
libssl3
libuuid
mdev-conf
musl
musl-utils
ncurses-terminfo-base
openrc
openssh
openssh-client-common
openssh-client-default
openssh-keygen
openssh-server
openssh-server-common
openssh-server-common-openrc
openssh-sftp-server
openssl
scanelf
ssl_client
zlib
myMPDos:~#
```

I do not have a serial cable, but I could grab another video of the boot process. Last time I did it I didn't notice anything weird, but I could have missed it!

stratosgear, 2024-04-12:
Well the video looks pretty clean.

https://drive.google.com/file/d/1d0FrXiKljYpj6HMtWRWCmJhMmrI1Rp5K/view?usp=sharing
(Github allows only up to 10MB uploads, This is about 20MB)

Apologies! The video is awful, low quality, and the webcam was keep trying to focus.  I can make it out though.  Plus, it is only a 1m 50secs long torture only!

I really apologize for the length of this thread but if YOU cannot shed some light I do not know who can help at this point!  I appreciate your patience and help on this matter!!!

jcorporation, 2024-04-13:
It seems the overlay is not loaded. Strange this is not specific to myMPD but alpine Linux standard.

I will try to reproduce this.

jcorporation, 2024-04-20:
I can reproduce your issue. The apk cache is not setup correctly. Root cause will be fixed in next release.

There is a workaround: type `apk cache -v sync` and reboot.

stratosgear, 2024-04-20:
Great! I verify that the workaround works.  

Is this a "permanent" workaround, or I have to keep redoing it before each reboot?

Of course, I would be upgrading to the latest version when the new release is out, but just so I know for the intermediate tests I will be doing.  Thanks!

jcorporation, 2024-04-20:
> Is this a "permanent" workaround, or I have to keep redoing it before each reboot?

No, run this once should fix the issue.

stratosgear, 2024-04-20:
I also have issues editing /etc/mpd.conf.  I make edits so I setup mpd in a satellite mode, but I cannot persist them.  After my edits (verified that I was able to change the file after doing a cat on the file) and executing a `savereboot.sh`, it seems that my changes are lost on the next reboot.

Is this related with this current sync issue...?

jcorporation, 2024-04-20:
`/etc/mpd.conf` is managed by the script `/usr/bin/configmpd.sh` as described in https://jcorporation.github.io/myMPDos/advanced-topics/customize-mpd.

stratosgear, 2024-04-20:
First of all sorry for bombarding you with all these issues, and I really appreciate for taking the time  to be so patient with me (although, I think I have indeed found some real bugs, and not everything is irrelevant chatter :) )

I am trying to collect information about the intended usage of myMPDOS that seems to be spread in multiple places.  Maybe I am missing a clear context here, that I have not seen explicitly explain in the docs.

Is the intended and proposed usage to preconfigure everything at sdcard creation only?

Or I am, somehow, allowed to continuously editing the configuration files at runtime and persist them as I go?

My current issue here is that I am not sure what the EXACT configuration of mpd and mympd should be in a satellite configuration (I know about the satellite page in the docs, and I am using it as a guide), so I am doing multiple edits until I get everything working.  Obviously, making random edits and re-burning an sdcard each time I want to make a change in a file, is TOO time consuming.  So, basically I want to keep doing changes in a live system, maybe do some reboots, and once I get all the parameters right,  then maybe do one final burn with all the right settings (and safekeep all the changes in case I want to reburn a new sdcard in the future)

So, under this context, it is not clear how I can use the info under the Customize Mpd section on the myMPDOS docs page.

 Thanks!

jcorporation, 2024-04-20:
To find a valid mpd config you can edit the `/etc/mpd.conf` file until it works. Before you reboot/shutdown copy it to `/etc/mympdos/custom/mpd.conf`  and save the changes with `lbu_commit`. Hope this helps.

closed 2024-04-20 by jcorporation