jcorporation/myMPDos #40: How to persist arbitrary files

stratosgear, 2024-04-20:
My usecase requires running tailscale for connecting my pi3 to my network (using vpn as keyword here, for anyone having issues with vpn access and similar issues)

Installation, configuration and usage of tailscale works correctly.

But the process of loging in to tailscale does not persist across reboots.  This is because tailscale keeps it's login state in `/var/lib/tailscale/tailscaled.state` (Note: to be verified, as I cannot test this across boots)

How would I be able to persist this file so it is available across reboots? 

Sidenote: I am vaguely familiar with overlays, and I am struggling to understand the exact mechanism that myMPDos is using for the alpine overlays.  Where can I read more about it?

Thanks!

jcorporation, 2024-04-20:
myMPDos uses the standard lib utility for persistence: https://wiki.alpinelinux.org/wiki/Alpine_local_backup

Simply use `lbu include` to include the desired directories in the lbu backup.

stratosgear, 2024-04-20:
Confirm this works!  You're the best!

closed 2024-04-20 by stratosgear