jcorporation/myMPDos #44: On First boot no myMPDos apks are installed

keithmillar, 2024-12-11:
The Boot log shows some failures one being unable to select packages. I wood like to get this working as it is a fantastic program.

myMPDos 1.6.0 bootstrap script

 *   WARNING: you are stopping a sysinit service
 *   Executing: /usr/libexec/rc/sh/openrc-run.sh /usr/libexec/rc/sh/openrc-run.sh /etc/init.d/modloop stop
 *   Unmounting /.modloop ... [ ok ]
Creating data partition on sd card

Command (m for help): The number of cylinders for this disk is set to 223290.
There is nothing wrong with that, but this is larger than 1024,
and could in certain setups cause problems with:
1) software that runs at boot time (e.g., old versions of LILO)
2) booting and partitioning software from other OSs
   (e.g., DOS FDISK, OS/2 FDISK)

Partition type
   p   primary partition (1-4)
   e   extended
Partition number (1-4): First sector (1048576-62521343, default 1048576): Using default value 1048576
Last sector or +size{,K,M,G,T} (1048576-62521343, default 62521343): Command (m for help): Using default value 62521343

The partition table has been altered.
Calling ioctl() to re-read partition table
Waiting for /dev/mmcblk0p1 found with FS
 *   Executing: /usr/libexec/rc/sh/openrc-run.sh /usr/libexec/rc/sh/openrc-run.sh /etc/init.d/modloop start
 *   Mounting modloop  ...
 *   Verifying modloop
 [ ok ]
Waiting for /dev/mmcblk0p2 found with NOFS
Formating data partition on sd card
(1/6) Installing libeconf (0.6.3-r0)
(2/6) Installing libblkid (2.40.2-r4)
(3/6) Installing libcom_err (1.47.1-r1)
(4/6) Installing e2fsprogs-libs (1.47.1-r1)
(5/6) Installing libuuid (2.40.2-r4)
(6/6) Installing e2fsprogs (1.47.1-r1)
Executing busybox-1.37.0-r8.trigger
OK: 14 MiB in 31 packages
mke2fs 1.47.1 (20-May-2024)
Discarding device blocks: done
Creating filesystem with 7684096 4k blocks and 1921360 inodes
Filesystem UUID: fbbc817a-c403-4411-929e-cf1c83b586d0
Superblock backups stored on blocks:
	32768, 98304, 163840, 229376, 294912, 819200, 884736, 1605632, 2654208,
	4096000

Allocating group tables: done
Writing inode tables: done
Creating journal (32768 blocks): done
Writing superblocks and filesystem accounting information: done

Mounting data partition on sd card

Filesystem usage
Filesystem                Size      Used Available Use% Mounted on
devtmpfs                 10.0M         0     10.0M   0% /dev
shm                     453.7M         0    453.7M   0% /dev/shm
tmpfs                   453.7M     15.7M    438.0M   3% /
tmpfs                   181.5M     48.0K    181.4M   0% /run
/dev/mmcblk0p1          511.7M     89.1M    422.6M  17% /media/mmcblk0p1
/dev/loop0               32.9M     32.9M         0 100% /.modloop
/dev/mmcblk0p2           28.7G      2.0M     27.2G   0% /media/mmcblk0p2

Mounts
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,relatime)
devtmpfs on /dev type devtmpfs (rw,nosuid,noexec,relatime,size=10240k,nr_inodes=106970,mode=755)
proc on /proc type proc (rw,nosuid,nodev,noexec,relatime)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000)
shm on /dev/shm type tmpfs (rw,nosuid,nodev,noexec,relatime)
[... 176 lines cut ...]
etc/zoneinfo/UTC
etc/localtime
etc/ssh/sshd_config
etc/ssh/ssh_host_rsa_key.pub
etc/ssh/ssh_host_rsa_key
etc/ssh/ssh_host_ecdsa_key.pub
etc/ssh/ssh_host_ecdsa_key
etc/ssh/ssh_host_ed25519_key.pub
etc/ssh/ssh_host_ed25519_key
etc/keymap/
etc/keymap/us.bmap.gz
etc/mympdos/
etc/mympdos/custom/
etc/mympdos/custom/mpd.replace
etc/passwd-
etc/shadow-
etc/group-
var/lib/misc/openrc-shutdowntime
var/lib/mympd/
var/lib/mympd/scripts/
var/lib/mympd/scripts/RW-Library.lua
var/lib/mympd/scripts/Reboot.lua
var/lib/mympd/scripts/Shutdown.lua
Created /media/mmcblk0p2/myMPDos.apkovl.tar.gz
Successfully saved apk overlay files

jcorporation, 2024-12-11:
Thanks for reporting this issue.

The mympdos-base package is not in the repository. I added it again, it was a build failure with Alpine Linux 3.21.

I also released a 1.6.1 release fixing the regressions introduces by the upgrade to Alpine Linux 3.21.

keithmillar, 2024-12-12:
Thanks For the fast update.
The software now appears to be installed, however no sound cards are installed so I can only play back in a browser and not locally on the pi.

closed 2024-12-12 by jcorporation