jcorporation/myMPDos #47: Lbu message on almost every saveshutdown or savereboot

pitsi, 2025-05-05:
I am a new user of mympdos, with some experience on alpine (x64, in vm and with "system disk install"). I have read about saveshutdown and savereboot and i do use them instead of poweroff and reboot.

While mpd and mympd work fine, I am having this issue with the rest of the system. When I run safereboot or safeshutdown, I get this message
```
The following apkovl file(s) were found:
/media/mmcblk0p2/mympdos.apkovl.tar.gz

Please use -d to replace.
```
I do not know what caused it in the first place, but I assume it was that change to enable the built in sound card (and it's module) as I said in the other issue and I do not know how to make it stop completely.
All I know is that it goes away until I run `lbu commit -d`, until it shows up again after a few reboots. And when that message shows up, nothing from what I changed is stored, so I have to do it again on the next boot. E.g. I may install htop, run savereboot, have that message pop up and after the reboot htop is not there.

So what did (or keep doing) wrong?

pitsi, 2025-05-06:
Minor question, probably irrelevant: how do I update the kernel on it? I thought it had linux-lts installed, but apk says there is no linux-* package installed at all!

pitsi, 2025-05-13:
I am trying to make a similar installation on virtualbox (and on x64) to find out how it works, so I have a question. Is it a diskless installation with /dev/mmcblk0p2 being the one that data is stored in?

jcorporation, 2025-05-13:
Yes. It is all standard and not myMPDos specific.

jcorporation, 2025-05-13:
> `lbu commit -d`

I wonder if it is save to use `lbu commit -d` always in the `saveshutdown.sh` and `savereboot.sh` scripts.

> Minor question, probably irrelevant: how do I update the kernel on it? I thought it had linux-lts installed, but apk says there is no linux-* package installed at all!

The kernel is in the image for the Raspberry Pi and not installed with apk. This is why you should use `mympdos-update.sh`. This script updates the base Raspberry Pi Image and alll apks. I release for each Alpine Linux release a myMPDos release.

pitsi, 2025-05-15:
Sorry for the delay in answering. First and foremost, I can not replicate a similar installation for x64 :(
Other than that, I just noticed that apk update says `3.21` as the distro's version, while I was used to `edge`, so that is the reason I have an older version of the kernel for instance. I stopped installing packages a few days after I opened the issue, so the system "just works" now.

commit 58353655 (2025-05-15): Fix: Use "lbu commit -d"

Closes #47
closed 2025-05-15 by jcorporation